beckdac/ESP8266-transparent-bridge #5: Feature request: Support different data bit, parity, stop bit

petgit, 2015-01-03:
The transparent-bridge firmware works fine! Thanks for the great project.
I want it use with a electronic meter MT174:  http://www.iskraemeco.si/emecoweb/eng/products/industrial_electronic_meters.html . The infrared serial port is configured to 300 Baud, 7 data bit, 1 stop bit even parity.

Currently the transparent-bridge firmware only supports  UART Mode 8 databit, 1 Stopp bit, No partity.

To get it working I modified a C# Terminal Client Example, which calculate the even parity bit before send the data and delete bit 8 on receive.

Also the most Viessmann boilers have an IR interface. But this use: 
4800 bps, 8 Bits, Even Parity and 2 Stopbits (4800,8,E,2) 
Possible the BAUD command can extended to support also 
+++AT BAUD 300,7,E,1
+++AT BAUD 4800,8,E,2

petgit, 2015-01-13:
I have implemented 
+++AT BAUD baudrate [databits [parity [stopbits]]]
with 
databits = 5/7/8, default 8
parity = N/E/O, default N
stopbits = 1/1.5/2, default 1
Example:
+++AT BAUD 300 7 E 1
OK
+++AT BAUD
BAUD=300 7 E 1

But there is an other problem: chars are missing. If I send some chars with HTERM only the first char is  received in Putty. And also chars from the meter output are missing. 
In the recvTask every char is read out from the fifo and send via TCP:

	while (READ_PERI_REG(UART_STATUS(UART0)) & (UART_RXFIFO_CNT << UART_RXFIFO_CNT_S))
	{
		WRITE_PERI_REG(0X60000914, 0x73); //WTD
		c = READ_PERI_REG(UART_FIFO(UART0)) & 0xFF;

		for (i = 0; i < MAX_CONN; ++i) {
			if (connData[i].conn) {
				espconn_sent(connData[i].conn, &c, 1);
			}
		}
	}

Possible better to read out all from fifo in a buffer and then send the whole buffer?

What mean: WRITE_PERI_REG(0X60000914, 0x73); //WTD  ?
Where is the documentation for the adress range 0X60000900 ?

petgit, 2015-01-25:
I made a pull request with +++AT BAUD baudrate [databits [parity [stopbits]]] and buffered write to wait for espconn_sent_callback of the pre-packet.

closed 2015-01-25 by petgit

beckdac, 2015-01-30:
Thanks again for your terrific work.  The pull has been merged.

probonopd, 2015-10-31:
@petgit were you able to use this with the Viessmann Optolink interface? If so, what is your circuit?

petgit, 2015-10-31:
I used the circuit from http://openv.wikispaces.com/Bauanleitung+LAN-Ethernet. Instead Lantronix I used a Raspberry Pi for an Optolink.  And the same for a Power meter with an ESP05.