beckdac/ESP8266-transparent-bridge #7: Change Setup and Data ports

petgit, 2015-01-15:
Now the data from the serial port and the configuration commands use the same port.
I think it is a good idea to separate this: i.e.
Telnet Port 23 fix for the AT commands.
And also possible for standard/warning/error output. 
The os_sprintf data must recieved with an extra serial input at GPIO 2.

AT PORT set the port for the serial data (i.e 10001 as in the XPORT).
Then it would simpler to use also UDP , i.e.
AT PORT <NR> [TCP|UDP] 

And also a client mode: The modul connects to a server or send UDP broadcasts.

monteslu, 2015-04-25:
+1

The current setup could be problematic if the bytes meant to be relayed to serial accidentally form a `+++AT` command

fvpalha, 2015-05-29:
Hi @petgit and @monteslu .

I changed the config tcp port - [fvpalha/ESP8266-transparent-bridge](https://github.com/fvpalha/ESP8266-transparent-bridge)
Please, can you test?

beckdac, 2015-06-08:
Thanks @fvpalha, I see your pull request and I'll try to test it as soon as I get new modules in sometime in the next week.  This is great!

Cougar, 2015-08-01:
This commit is full of different changes in addition to second TCP port. @fvpalha could you separate new feature from other changes? The current patch is hard to read read and understand but also can't be simply merged with upstream code.