beckdac/ESP8266-transparent-bridge #8: AT command from local uart

djlover, 2015-01-15:
Good day to everyone, I'm trying to use +++AT command but on side uart ( not tcp side of communication ) this will able to change paramenter, join network from local uart device, but I'm still in trouble for now..

petgit, 2015-01-15:
The 'AT-Interpreter' is only on network side. I have lost the connection to a device by enter a none numeric value for n with +++AT MODE n. Which is interpreted as  +++AT MODE 0: no Station and no AP.  I have flashed  LUA  and  set the mode back from 0 to 1.
But an 'AT-Interpreter' at uart side would only help if the baudrate is known.
I think it would be better to make a condition at some GPIO at boot to reset the settings to  initial values as AP with known password. Similiar GPIO=0 at boot to enter flashmode. Any ideas? Some modules have limited GPIO connected. Possible check if a connection between TX & Rx or GPIO 0 exists, if yes, set defaults.

beckdac, 2015-05-05:
The AT command set is only on the network side. The GPIO2 is now exposed if that helps implement your idea petgit.

closed 2015-05-05 by beckdac