beckdac/ESP8266-transparent-bridge #12: Can't change the WiFi channel.

firew4lker, 2015-02-12:
I am not able to change the WiFi channel. I is always 6. The command I am using is

+++AT AP myssid 12345678 3 10

The software will respond OK, but the channel will not change. 

Can someone verify?

beckdac, 2015-02-12:
Did you reset the unit after the change?

https://github.com/beckdac/ESP8266-transparent-bridge/issues/12

firew4lker, 2015-02-13:
Yes. Reset, and power cycle.

firew4lker, 2015-02-13:
I will try a re-flash tomorow.

openSoftMan, 2015-02-14:
I have the same issue; channel doesn't change.  The command is accepted, but always stays at 6.

firew4lker, 2015-02-14:
Can the channel be changed while in mode 3 (AP and STA)? Maybe if it have to connect to an AP, it must transmit to the same channel?

firew4lker, 2015-02-14:
Disregard the above. I thought my home AP was at channel 6 also. But no (it actually 11).

firew4lker, 2015-02-15:
It seems the when in MODE 3 the AP channel is always the channel the STA is connected to.

beckdac, 2015-05-22:
@firew4lker yes, this is the SDK behavior.  Thanks for the notifications.

closed 2015-05-22 by beckdac