beckdac/ESP8266-transparent-bridge #13: Implementation of a "heartbeat".

firew4lker, 2015-02-15:
Since the code can't interpret AT command originating from the UART, what is the best way for a microcontroller to check if the module is alive or frozen?

Maybe a mode that GPIO -2 toggles every couple o seconds or so?

The best scenario would be UART AT commands support. Is it even possible to implement support of AT command from the UART side?

beckdac, 2015-12-02:
I see, so the unit attached to the ESP8266 wants to know if a destination is available?

firew4lker, 2015-12-04:
No, if the ESP8266 is ok. Eg. if it has frozen the microcontroller to be able to reset it.

beckdac, 2015-12-17:
OK.  How does the ESP8266 monitor the microcontroller?  Lack of serial activity then it throws the GPIO pin?

firew4lker, 2015-12-17:
I mean the opposite. The microcontroller to monitor the ESP8266. If for example the ESP8266 stops toggling a GPIO then the mcu will know that the module is stuck and should be restarted.

I have notice that it fairly common for the "ebay tyoe" modules to freeze.

beckdac, 2015-12-20:
OK.