beckdac/ESP8266-transparent-bridge #14: strange statement causing latency

tve, 2015-02-18:
Can you explain what the following line does:
```
WRITE_PERI_REG(0X60000914, 0x73); //WTD
```
from https://github.com/beckdac/ESP8266-transparent-bridge/blob/master/user/user_main.c#L39

It is necessary as far as I can tell, but there is something weird happening overall which causes a lot of latency and I can't find any description or mention of register 0X60000914 anywhere.

ArnieO, 2015-02-18:
According to the brief comment on the link below, it seems to be linked to the watchdog timer?
https://github.com/willemwouters/ESP8266/blob/master/sdk/esp_iot_sdk_v0.9.3/examples/at_v0.20_on_SDKv0.9.3/at/user/at_port.c

tve, 2015-02-18:
Ah, thanks I hadn't seen that version of the AT example. The next version, e.g. https://github.com/espressif/esp8266_at/blob/master/at/driver/uart.c doesn't have the WDT line in it anymore. The best I could find about the actual WDT registers is https://github.com/themadinventor/esp-blobless/blob/master/include/esp8266.h which indicates that this line resets the WDT. Why that would be needed here is beyond me, but what I observe is that the whole RX code construction introduces 200-300ms latency. Something's off...

tve, 2015-02-19:
I've partially resolved my issue, it's not the WDT or uart that is causing the latency but the sent callback that is very late. Now trying to find out why...

closed 2015-02-19 by tve