beckdac/ESP8266-transparent-bridge #15: Software Development Kit for ESP8266

fvpalha, 2015-04-17:
Hi.
Congratulations by the solution. It is really I need in my project.
I am a newbie in ESP8266 development.
Where I can find a Starter Tutorial?
Best regards.

fvpalha, 2015-04-20:
I am trying to use the Unofficial Development Kit for Espressif ESP8266 - http://www.esp8266.com/viewtopic.php?f=9&t=820.
Someone have a tip?

fvpalha, 2015-04-22:
Hi.
I changed the Makefile and now all is working.
Best regards.

beckdac, 2015-05-05:
Can you provide a diff for the Makefile change?

fvpalha, 2015-05-12:
Hi beckdac,

I just used a Makefile from another example by esp8266-devkit (https://github.com/CHERTS/esp8266-devkit/tree/master/Espressif/examples)

```
#############################################################
#
# Root Level Makefile
#
# (c) by CHERTS <sleuthhound@gmail.com>
#
#############################################################

BUILD_BASE	= build
FW_BASE		= firmware

# Base directory for the compiler
XTENSA_TOOLS_ROOT ?= c:/Espressif/xtensa-lx106-elf/bin

# base directory of the ESP8266 SDK package, absolute
SDK_BASE	?= c:/Espressif/ESP8266_SDK

# esptool path and port
SDK_TOOLS	?= c:/Espressif/utils
ESPTOOL		?= $(SDK_TOOLS)/esptool.exe
ESPPORT		?= COM1
ESPBAUD		?= 115200

# BOOT = none
# BOOT = old - boot_v1.1
# BOOT = new - boot_v1.3+
BOOT?=none
# APP = 0 - eagle.flash.bin + eagle.irom0text.bin
# APP = 1 - user1.bin
# APP = 2 - user2.bin
APP?=0
# SPI_SPEED = 20MHz, 26.7MHz, 40MHz, 80MHz
SPI_SPEED?=40
# SPI_MODE: QIO, QOUT, DIO, DOUT
SPI_MODE?=QIO
# SPI_SIZE: 256KB, 512KB, 1024KB, 2048KB, 4096KB
SPI_SIZE?=512

ifeq ($(BOOT), new)
    boot = new
else
    ifeq ($(BOOT), old)
        boot = old
    else
        boot = none
    endif
endif

ifeq ($(APP), 1)
    app = 1
else
    ifeq ($(APP), 2)
        app = 2
    else
        app = 0
    endif
endif

ifeq ($(SPI_SPEED), 26.7)
    freqdiv = 1
	flashimageoptions = -ff 26m
else
    ifeq ($(SPI_SPEED), 20)
        freqdiv = 2
        flashimageoptions = -ff 20m
[... 251 lines cut ...]
else
	$(ESPTOOL) -p $(ESPPORT) -b $(ESPBAUD) write_flash $(flashimageoptions) $(addr) $(FW_BASE)/upgrade/$(BIN_NAME).bin
endif
endif

flashinit:
	$(vecho) "Flash init data:"
	$(vecho) "Clear old settings (EEP area):"
	$(vecho) "clear_eep.bin-------->0x79000"
	$(vecho) "Default config (Clear SDK settings):"
	$(vecho) "blank.bin-------->0x7E000"
	$(vecho) "esp_init_data_default.bin-------->0x7C000"
	$(ESPTOOL) -p $(ESPPORT) -b $(ESPBAUD) write_flash $(flashimageoptions) 0x79000 $(SDK_BASE)/bin/clear_eep.bin 0x7c000 $(SDK_BASE)/bin/esp_init_data_default.bin 0x7e000 $(SDK_BASE)/bin/blank.bin

rebuild: clean all

clean:
	$(Q) rm -f $(APP_AR)
	$(Q) rm -f $(TARGET_OUT)
	$(Q) rm -rf $(BUILD_DIR)
	$(Q) rm -rf $(BUILD_BASE)
	$(Q) rm -rf $(FW_BASE)

$(foreach bdir,$(BUILD_DIR),$(eval $(call compile-objects,$(bdir))))
```

beckdac, 2015-05-22:
Does that Makefile work for you?

closed 2015-05-22 by beckdac

fvpalha, 2015-05-22:
Yes @beckdac. Do you have another?

beckdac, 2015-05-24:
No.  Yours worked on one box.  The repo version worked on another.  Many thanks for posting yours.