beckdac/ESP8266-transparent-bridge #18: Initial conditions

PA7T, 2015-04-27:
Dear beckdac,

your project is exactly what I was looking for :+1: 
Unfortunately, I do not have windows and therefore cannot compile your code :(
After cloning I successfully uploaded the firmware, but cannot connect to the device.
As far as I understand, the supplied binaries are for the dynamic configuration.
In my case the ESP8266 does not open any AP, so I have no change of sending any AT commands via TCP.

Would it be possible to include a binary version in the repo that just opens an default AP, takes a default IP and port and start listening?

Cheers,

PA7T

ArnieO, 2015-04-27:
PA7T: The binaries are there already:
https://github.com/beckdac/ESP8266-transparent-bridge/tree/master/bin
It works for me (on Windows 7). 
Remember you need to use telnet to use the +++AT commands.

PA7T, 2015-04-30:
hmm I uploaded them successfully, but I cannot find an AP on the air in the end. so no chance to use telnet...

Heavy-Horses, 2015-05-01:
I'm having the exact same issue with the prebuilt bins. Uploaded with nodemcu flasher windows gui and with the python based esptool.py and i get the same results ether way.

eriksl, 2015-05-01:
Over here it works, although for some obscure reason I had to do the cycle twice on one esp8266 (flash normal firmware, configure ap, flash bridge firmware).

Heavy-Horses, 2015-05-02:
I found my problem. it appears i had a corrupted download of "eagle.app.v6.irom0text.bin" the md5 should be   E812D4D75B9C8809805995991E9DEC13 mine was D391F00B4B4861AC9B0BFE191B48BA1D
This time I downloaded the complete ziped project, then extracted the bins and flashed it and it worked.
Remember a zip file gives you another level of error checking.

PA7T, 2015-05-02:
in my case the MD5 is fine and I flash with:
sudo esptool.py --port /dev/tty.usbserial-A6* write_flash 0x00000 eagle.app.v6.flash.bin 0x40000 eagle.app.v6.irom0text.bin
power cycling does not help. I just do not see anything on the air :(

eriksl, 2015-05-03:
Did you try my workaround?

eriksl, 2015-05-03:
BTW the version of jvdbroek I didn't get working, no reaction of the device whatsoever.

beckdac, 2015-05-22:
I just went through this myself setting up new devices.  I completely borked some of the modes.  After doing this, I reflashed the stock AT firmware paying particular attention to the esp_init and blank.bins, e.g. on this page: https://github.com/espressif/esp8266_at/wiki/Compiler

closed 2015-05-22 by beckdac