beckdac/ESP8266-transparent-bridge #19: Can't change PORT

denis5000, 2015-04-29:
Hello.
This is a very good firmware and this is exactly what every arduino user needs.

I have started playing with it and have a problem that the PORT can't be changed.
First it started on port 23 and when I sent it a command "+++AT PORT 80" the ESP rebooted and opened port 16 (instead of 23 or 80).
I have tried but can't change it to any other port.

Please help.

beckdac, 2015-05-05:
Hi denis, I can try to test this tonight.  I recall testing this during development and it was successful.

beckdac, 2015-05-19:
I cannot confirm this.  When I changed to port 80, the connection drops and all further connections work on port 80.  I also tried doing a reset in between changing the port and reconnecting.

closed 2015-05-19 by beckdac