beckdac/ESP8266-transparent-bridge #21: would like to have the STA address broadcasted via MDNS

monteslu, 2015-05-01:


beckdac, 2015-05-05:
Can you describe the use case?

monteslu, 2015-05-06:
Currently it is difficult to know what the dynamic IP of your device is.  If an MDNS broadcast was being sent out, apps that wish to connect to the device could find it by service name.

Things like Chromecast, and wireless printers, and other headless devices commonly use this pattern.

beckdac, 2015-05-06:
Are you aware of any other ESP8266 implementations I could grab code from for this?

monteslu, 2015-05-06:
This might work:  https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/libraries/ESP8266mDNS/examples/mDNS_Web_Server/mDNS_Web_Server.ino

beckdac, 2015-12-02:
Not the first time this has been useful.