beckdac/ESP8266-transparent-bridge #24: No new Wifi afte flashing

sigrokBlack, 2015-05-14:
Hi,
after successfully flashing your firmware and power cycling, no wifi networks appears on my Pc/Android.

What can I do?

Greetings Joshua

sigrokBlack, 2015-05-14:
Sloved It! When you used NODEMCU before on the module do this :+1: http://www.rei-labs.net/esp8266-update-firmware-to-change-the-baudrate/1/

closed 2015-05-14 by sigrokBlack

MustafaFayez, 2015-08-16:
@sigrokBlack  I have the same problem: after flashing transparent-bridge my pc/tablet can't see the module's AP!! I tried to flash the basic AT firmware first and config wifi AP and then flash the transparent-bridge firmware again, it didn't work. How you solved it?

sigrokBlack, 2015-08-17:
Hi, in my last post, there is a link with detailed instructions. The main problem is that firmwares like nodemcu are saving their files all over the flash, so you have to earse the flash chip completly.
Regards Joshua

beckdac, 2015-12-17:
Yes. I can confirm this.