beckdac/ESP8266-transparent-bridge #25: No "net" option in avrdude

greenrider, 2015-05-25:
Hello.
I am trying to upload a sketch to arduino via avrdude (on Windows OS) and I can't find avrdude that will support the "net" option.
Can you please give an example how to do it?

Thank you.

beckdac, 2015-05-25:
Sure... this is the command I use for an avr running optiboot connected to the esp8266:

host=192.168.1.20
hex=mysketch.hex
avrdude -v -v -c avrisp -p m328p -P net:$host:23 -F -U flash:w:$hex:i

I highly recommend using GPIO2 pin as the arduino reset.  Then you can do it entirely wirelessly.

So the net is the value for the -P (port) option.  Is it possible that Windows does not support a socket port?

greenrider, 2015-05-25:
I have tryed that, but windows doesn't support this feature.
My main goal is beeing able to upload a new sketch to Arduino via an Android App that I am writing.
Do you use linux for uploading?
Maybe you know how to do it via java on Android? (I may be asking too much) :)

beckdac, 2015-05-26:
I use OS/X and linux.  On OS/X, I use the macports avrdude.  On linux, I use Ubuntu 14.04 and use the standard repo avrdude.

Unfortunately, I don't know about windows.  A google search didn't turn up anything revealing with: https://www.google.com/webhp?sourceid=chrome-instant&ion=1&espv=2&ie=UTF-8#q=windows+avrdude+-P+net

Again, I don't know about Android.  I do know that it is possible to do this with a virtual machine under windows.  Just as it is possible to build the esp firmware there.  There are a bunch of esp toolchains in dockerhub.

closed 2015-12-02 by beckdac