beckdac/ESP8266-transparent-bridge #28: Compile failed on Windows

PanJ, 2015-06-20:
Makefile_VS2013 does not work on my Windows 8 environment. The error is

```
FW bin/eagle.app.v6.flash.bin
         usage: esptool [-h] [--port PORT] [--baud BAUD]
                        {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,flash_id,read_flash,erase_flash}
                        ...
     1>esptool : error : argument operation: invalid choice: 'build/app.out' (choose from 'load_ram', 'dump_mem', 'read_mem', 'write_mem', 'write_flash', 'run', 'image_info', 'make_image', 'elf2image', 'read_mac', 'flash_id', 'read_flash', 'erase_flash')
         make: *** [bin/eagle.app.v6.flash.bin] Error 2
```

I am trying to trackdown `FW` but cannot find any `FW` variable initialization. I believe that it has something to do with `FW_TOOL`. It seems that the command is not being executed correctly.

Maybe this is related with #15 but his makefile does not work for me.

beckdac, 2015-06-26:
Hi,
This looks like it is failing to do the upload, not the build.  Are the bins in the the bin directory?  Could you try uploading then with esptool yourself?

PanJ, 2015-06-29:
I am able to solve the issue by installing both esptool and esptool.py. They are both needed for building the project.

closed 2015-06-29 by PanJ