beckdac/ESP8266-transparent-bridge #29: Question: Data Rates & Overclock & Packet Loss

omersiar, 2015-06-25:
Hello, thank you for your firmware, it does really simplify the job. I am trying to use ESP8266 as telemetry module on ArduPilot 2.6. It connects on telemetry port and have 57600 baud rate. I flashed your firmware to ESP8266 and made necessary changes like baud rate, TCP port, AP settings. Everything seems working, I can connect to ESP8266, and can open the TCP port, and can see data. But some small amount of packets are being lost everytime and when client requests more data, lost packet rate increases dramatically. I did not have any problem with HC-05 Bluetooth UART module which is Bluetooth 2 + EDR module and has 2Mbps and 3Mbps data rates. 

So here is my questions ?

Do you think that packet loss is about data rate?
Can we overclock your firmware in order to achieve higher data rates?
Data streaming from ArduPilot is MAVlink protocol packets and non ASCII chars, does this affect packet loss?

beckdac, 2015-06-26:
I think the packet loss is about very small buffers on the arduino and no real buffer on the esp8266 firmware.  There was a fork that you can look for where the author added some buffers for both input and output, as I recall.  

I don't think clocking is the issue as much as characters getting dropped due to a lack of buffering.

I have done binary protocols with this firmware with a problem, however you might want to make sure there are no +++AT bytes being generated as part of the protocol.  This is unlikely, but it would cause the firmware to go into a command processing mode and not pass some characters.  You can also try the separate configuration firmware fork if you believe this is the problem.  Hope this helps.

omersiar, 2015-10-07:
Hello, Ive done some tests, it ends up ESP does not respond after awhile. +++AT command is not the problem. Same with Issue #1, I guess such application needs more buffer, real one.

closed 2018-11-17 by omersiar