beckdac/ESP8266-transparent-bridge #30: question: Static settings

M3gaFr3ak, 2015-07-24:
First: How do i set the baudrate for static settings?
Second: Do static settings completely deactivate the parsing of +++AT commands?

Cougar, 2015-08-01:
1. Defaults are in https://github.com/beckdac/ESP8266-transparent-bridge/blob/master/user/flash_param.c#L48

2. Yes. Look at https://github.com/beckdac/ESP8266-transparent-bridge/blob/master/user/server.c#L102-L106

M3gaFr3ak, 2015-08-04:
thanks!

another little question:
Currently i am using that as a bridge for controlling an led cube over wifi. but it has a delay of ~half a second. is that some sort of computation delay inside the esp?

beckdac, 2015-08-11:
Nope.  That is particularly long.  When you ping the device what is your roundtrip time?  I rarely see more than 10s of milliseconds.

closed 2015-12-02 by beckdac