beckdac/ESP8266-transparent-bridge #32: Static IP?

arlukas, 2015-08-11:
First of all thanks for the effort and for this really nice code.
I wonder if it's possible to define a static ip for the station mode. I've not seen it on the instructions.
If it's possible, how can I define it?

Thanks in advance.
Regards.

beckdac, 2015-11-01:
Does #define CONFIG_STATIC do it for you?

arlukas, 2015-11-04:
Hi.
Sorry, but I wasn't specific in my question.
I meant if I can define a static IP through the AT mode interface that you've implemented. Something like this: +++AT STATICIP.

Thanks in advance.

Regards.

beckdac, 2015-12-02:
Ah.  I can see the use case for that.  To eliminate the need for this, on my IoT networks, I have the MAC address of these devices in a table of 'static' address on the DHCP server.

bstott, 2015-12-20:
+1  This seems a vital component/feature. Many networks we create with this device do not need to be public and should not require a DHCP server.

pendru, 2016-04-09:
First of all thanks for a simple & smart solution.
looking forward for a STATIC IP in STA Mode, which brings the support for most of the use cases.

Thanks in advance

rajibgupta, 2016-04-09:
Great firmware,It saves a lot of time. But it would be great if we could provide a static IP to it.