beckdac/ESP8266-transparent-bridge #34: How to connect to bridge the first time to join it to an AP?

cburgess129, 2015-09-15:
How do I connect to the bridge the first time to connect it to an AP?  
Does it automatically join the nearest open AP?  
Do you set the default AP and password in the config?  
If it is not connected to an AP, it doesn't have an IP so you can't telnet into it.  Do the +++AT commands work via the UART or only via telnet?

eriksl, 2015-09-15:
Flash an AT-style firmware, configure the WLAN using AT-commands (AT+CWJAP etc.), then flash the bridge firmware and it will connect automatically.

cburgess129, 2015-09-15:
That is funny because that is what I did (by accident) the first time.  I didn't know that was the actual procedure.  Thanks

cburgess@netapp.com.com
614.352.8751

beckdac, 2015-12-02:
Thanks @eriksl & @cburgess129 : do either of you want to take a stab at modifying the README.md to reflect your experiences?  That would be very useful to others.

Testato, 2015-12-29:
so this is the offcial way ?

I thinks will be better if the firmware start in AP mode, whithout password. So the user can search it and connect directly to it. After he can configure his connection to his router if he want.

beckdac, 2016-04-29:
It isn't ideal, but yes, it is what I do for new chips.

closed 2016-04-29 by beckdac