beckdac/ESP8266-transparent-bridge #36: Error on compile.

Duality4Y, 2015-09-22:
Hi I get a Error when I try and compile the code.

this is the output:
```
robert@laptop:~/ESP8266-transparent-bridge$ make
CC driver/uart.c
CC user/flash_param.c
CC user/config.c
CC user/user_main.c
CC user/server.c
AR build/app_app.a
LD build/app.out
FW firmware/0x00000.bin
usage: esptool [-h] [--port PORT] [--baud BAUD]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,flash_id,read_flash,erase_flash}
               ...
esptool: error: argument operation: invalid choice: 'build/app.out' (choose from 'load_ram', 'dump_mem', 'read_mem', 'write_mem', 'write_flash', 'run', 'image_info', 'make_image', 'elf2image', 'read_mac', 'flash_id', 'read_flash', 'erase_flash')
Makefile:109: recipe for target 'firmware/0x00000.bin' failed
make: *** [firmware/0x00000.bin] Error 2
```

any ideas?
this is my makefile:
http://pastebin.com/kyHVRhM8

atomiclama, 2015-10-01:
I guess you are using the latest version of esptool which looks like it has changed it's command line options. 
I have this problem at the moment as well, I'm looking into it will let you know what I find

atomiclama, 2015-10-01:
Ok I modified the make file to add a new target as follows, and the image now uses to correct option to create 2 bin images. Enjoy if you see this

all: checkdirs $(TARGET_OUT) image
	
image: $(TARGET_OUT)
	$(vecho) "FW $@"
	$(Q) $(FW_TOOL) elf2image $(TARGET_OUT)

Duality4Y, 2015-10-03:
Ok I added it and ran it, but it gives me a syntax error.
```robert@laptop:~/ESP8266-transparent-bridge$ make clean
robert@laptop:~/ESP8266-transparent-bridge$ make all
CC driver/uart.c
CC user/flash_param.c
CC user/config.c
CC user/user_main.c
CC user/server.c
AR build/app_app.a
LD build/app.out
FW firmware/0x00000.bin
usage: esptool [-h] [--port PORT] [--baud BAUD]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,flash_id,read_flash,erase_flash}
               ...
esptool: error: argument operation: invalid choice: 'build/app.out' (choose from 'load_ram', 'dump_mem', 'read_mem', 'write_mem', 'write_flash', 'run', 'image_info', 'make_image', 'elf2image', 'read_mac', 'flash_id', 'read_flash', 'erase_flash')
Makefile:111: recipe for target 'firmware/0x00000.bin' failed
make: *** [firmware/0x00000.bin] Error 2
robert@laptop:~/ESP8266-transparent-bridge$ make image 
build/app.out
build/app.out: 1: build/app.out: Syntax error: "(" unexpected
Makefile:151: recipe for target 'image' failed
make: *** [image] Error 2

```

atomiclama, 2015-10-05:
@Duality4Y  paste in your make file and I'll have a look, I just quickly threw it together without worrying about the existing targets. It looks like it is still trying todo the old esptool command. I'm currently reworking this project to better fit my requirements. As not sure if this project is active. I'll let you when it is stable if you are interested.

Duality4Y, 2015-10-05:
here is a pastebin of the makefile:
http://pastebin.com/UeJekXD5

atomiclama, 2015-10-06:
You need to change line 108 to match 

all: checkdirs $(TARGET_OUT) **image**

So that the new target is used not the old ones. **image** is the new target that uses the new esptool options

Duality4Y, 2015-10-08:
now I get the following error:
```
robert@laptop:~/ESP8266-transparent-bridge$ make all
CC driver/uart.c
CC user/flash_param.c
CC user/config.c
CC user/user_main.c
CC user/server.c
AR build/app_app.a
LD build/app.out
build/app.out
build/app.out: 1: build/app.out: Syntax error: "(" unexpected
Makefile:151: recipe for target 'image' failed
make: *** [image] Error 2
```

Duality4Y, 2015-10-08:
Ok got it to compile ! and make the binaries !
I had to change this:
```
image: 
        $(TARGET_OUT)
	$(vecho) "FW $@"
	$(Q) $(FW_TOOL) elf2image $(TARGET_OUT)
```
into this:
```
image: $(TARGET_OUT)
	$(vecho) "FW $@"
	$(Q) $(FW_TOOL) elf2image $(TARGET_OUT)

```
and then it ran without any errors.
Thank you so much for  your help!

closed 2015-10-08 by Duality4Y

atomiclama, 2015-10-09:
Nice one glad I was able to help.

ankitmcgill, 2016-04-26:
can you please post the complete makefile? I am getting the same errors due to incorrect argument format

Duality4Y, 2016-04-28:
Sorry Don't have the makefile anymore.