beckdac/ESP8266-transparent-bridge #45: Uploading program to arduino wirelessly

rajibgupta, 2016-04-09:
Hello,
It isn't quite clear in the post about how to perform the wireless program upload to a microcontroller.
Can you please help me a bit in how to go about it ?

dima-ch, 2016-04-10:
Hello, rajibgupta
1. Setup USAT bootloader (https://github.com/Optiboot/optiboot) or use arduino.
2. Connect esp8266 to microcontroller or arduino:
RX--->TX
TX--->RX
VCC--->VCC = 3.3V
GND--->GND
Make sure that the power voltage of microcontroller equals 3.3V.
3. Reset microcontroller and upload your sketch using avrdude's socket port, e.g.
avrdude -c avrisp -p m328p -P net:192.168.4.1:23 -F -U flash:w:mySketch.hex:i
while LED of bootloader is flashing

closed 2016-04-16 by beckdac

rajibgupta, 2016-04-16:
If we have to reset the controller manually, then what's the point of wirelessly uploading?

beckdac, 2016-05-26:
You reset using telnet with the commands from the README.md:

`++++AT GPIO2 <0|1|2 100>                  # 1: pull GPIO2 pin up (HIGH) 0: pull GPIO2 pin down (LOW) 2: reset GPIO2, where 100 is optional to specify reset delay time in ms (default 100ms)`

Connect the ESP GPIO2 to your AVR RESET pin.