beckdac/ESP8266-transparent-bridge #53: AT command to set Tx RF power level

nbonniere, 2017-03-20:
For a short range point to point WiFi serial communication, it would be nice to be able to reduce the Tx power.  It would also reduce power consumption on battery.

beckdac, 2017-04-16:
Is this possible with the ESP libraries?

nbonniere, 2017-04-16:
It seems to be possible to set the power from 0 to 20.5 according to some comments such as on:
https://github.com/esp8266/Arduino/issues/764
and
https://github.com/esp8266/Arduino/blob/41074245a8d3029270622a8b489d53fdfedb2a22/libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.cpp#L169-L179

/**  * set the output power of WiFi
 * @param dBm max: +20.5dBm  min: 0dBm  */
void ESP8266WiFiGenericClass::setOutputPower(float dBm) {
    if(dBm > 20.5) {
        dBm = 20.5;
    } else if(dBm < 0) {
        dBm = 0;
    }
    uint8_t val = (dBm*4.0f);
    system_phy_set_max_tpw(val);
}

nbonniere, 2017-04-16:
The datasheet for the AT command set also has power control support in 0.25 db steps, 0 to 20.5 : <TX power>  maximum value of RF TX power, range 0 ~ 82, unit:0.25dBm

https://www.itead.cc/wiki/images/5/53/Esp8266_at_instruction_set_en_v1.5.4_0.pdf